smartdevicelink/sdl_core #3668: SDL response with "GENERIC_ERROR" in case request ResetGlobalProperties ("USER_LOCATION")

VjKlepikov, 2021-03-18:
### Bug Report
SDL response with "GENERIC_ERROR" in case request ResetGlobalProperties ("USER_LOCATION")

#### Precondition:
    1. Start SDL and HMI.
    2. Connect mobile device via any transport.
    3. Mobile application is registered and activated.
    4. Mobile application successfully sends RC.SetGlobalProperties(params)
    SetGlobalProperties { "userLocation":{"grid": {"col":2, "level":2,"row":2, "colspan":1,"levelspan":1, "rowspan":1 }

##### Reproduction Steps
1. Send ResetGlobalProperties with "USER_LOCATION"
App→SDL: ResetGlobalProperties (USER_LOCATION)

##### Expected Behavior
SDL sends RC.SetGlobalProperties request with default driver seat location properties to HMI
SDL→HMI: RC.SetGlobalProperties (
{ "userLocation":{"grid":{"col":0, "level":0,"row":0, "colspan":1,"levelspan":1, "rowspan":1 }}} 
SDL→HMI: RC.SetGlobalProperties ( SUCCESS )
SDL→App: ResetGlobalProperties( "success":true, resultCode":"SUCCESS" )

##### Observed Behavior
SDL→App: ResetGlobalProperties( "success":false , "resultCode":"GENERIC_ERROR")

##### SDL_logs:
[SDL_Log.zip](https://github.com/smartdevicelink/sdl_core/files/6162715/SDL_Log.zip)

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (c9fc559)
* Testing Against: HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop (41c804d), SPT: SPT: 20210305-Android (1076)

VjKlepikov, 2021-03-18:
@yang1070 Could you please confirm the expected result: SDL sends RC.SetGlobalProperties request with default driver seat location properties to HMI RC.SetGlobalProperties (
{ "userLocation":{"grid":{"col":0, "level":0,"row":0, "colspan":1,"levelspan":1, "rowspan":1 }}}

yang1070, 2021-04-08:
@VjKlepikov According to the proposal https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0221-multiple-modules.md `The default userLocation should be DRIVER's seat to ensure backwards compatibility.`
Therefore, `grid":{"col":0, "level":0,"row":0, "colspan":1,"levelspan":1, "rowspan":1 }}` is correct

ShobhitAd, 2021-08-09:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3741

PR #3741 (2021-08-09): Fix handling of RC.SetGlobalProperties response
Fixes #3668 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Can be reproduced manually:
- Send ResetGlobalProperties with "USER_LOCATION"
- App→SDL: ResetGlobalProperties (USER_LOCATION)

**Expected Behavior**
SDL sends RC.SetGlobalProperties request with default driver seat location properties to HMI
SDL→HMI: RC.SetGlobalProperties (
{ "userLocation":{"grid":{"col":0, "level":0,"row":0, "colspan":1,"levelspan":1, "rowspan":1 }}}
HMI→SDL: RC.SetGlobalProperties ( SUCCESS )
SDL→App: ResetGlobalPrope [... cut]
closed 2021-08-09 by ShobhitAd