smartdevicelink/sdl_core #3674: Core sends OnAppRegistered before app has valid internal HMI state

jacobkeeler, 2021-03-24:
### Bug Report
Core sends a `BC.OnAppRegistered` notification to the HMI before setting up the internal HMI state for the app. If the HMI sends a notification (such as `OnSystemRequest`) to this app before the HMI state is set up, the app will not receive this notification.

##### Reproduction Steps
1. Set up your HMI to immediately send a notification to an app when `BC.OnAppRegistered` is received. `OnSystemRequest` will be used in this example, since it can be reproduced with the attached HMI PR on EXTERNAL_PROPRIETARY policy mode.
2. Register an app

##### Expected Behavior
App receives OnSystemRequest from Core

##### Observed Behavior
App does not receive OnSystemRequest from Core

##### OS & Version Information
* OS/Version: Ubuntu 20.04
* SDL Core Version: [release/7.1.0-RC1](https://github.com/smartdevicelink/sdl_core/tree/release/7.1.0-RC1) branch
* Testing Against: HMI - https://github.com/smartdevicelink/sdl_hmi/pull/540, App - SDL Test Suite Android

jacobkeeler, 2021-07-27:
closing with the merge of #3674

dboltovskyi, 2021-10-01:
@jacobkeeler Please notice we were able to reproduce this issue at https://github.com/smartdevicelink/sdl_core/releases/tag/8.0.0-RC1 (86e032f) using script [3674_SDL_sends_OnAppRegistered_before_app_has_valid_internal_HMI_state](https://github.com/smartdevicelink/sdl_core/files/7268902/3674_SDL_sends_OnAppRegistered_before_app_has_valid_internal_HMI_state.lua.zip)
Policy flow: any
Report and logs: [2021-10-01_12-53-24.336_3674.zip](https://github.com/smartdevicelink/sdl_core/files/7268926/2021-10-01_12-53-24.336_3674.zip)
Probability: ~20% so the script needs to be run 25-50 times in a row in parallel ATF mode.

Could you try to reproduce it on your end?

dboltovskyi, 2021-10-15:
@jacobkeeler Please notice we were able to reproduce this issue at [release/8.0.0](https://github.com/smartdevicelink/sdl_core/commits/release/8.0.0) (0ed38e8d2f7cb3a61a303cca2806a06ce9a1d58d)
Script: [test_scripts/Defects/8_0/3674_1_HMI_sends_OnSystemRequest_before_UpdateAppList.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/release/8.0.0/test_scripts/Defects/8_0/3674_1_HMI_sends_OnSystemRequest_before_UpdateAppList.lua)
Probability: 15%
Report and logs: [2021-10-15_13-24-32.585_3674.zip](https://github.com/smartdevicelink/sdl_core/files/7355088/2021-10-15_13-24-32.585_3674.zip)
Message in SDL log:
> WARN  [15 Oct 2021 17:24:40,823][140477223393024][Commands] /home/developer/sdl/sdl_core/src/components/application_manager/rpc_plugins/sdl_rpc_plugin/src/commands/hmi/on_system_request_notification.cc:101 commands::OnSystemRequestNotification::Run: Can't select application to forward OnSystemRequest.

iCollin, 2021-11-09:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3806

PR #3675 (2021-07-27): Suspend notifications to mobile until HMI state is established
Fixes #3674 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Run full ATF regression to verify changes do not break anything.
Run manual test of sending OnSystemRequest from HMI before app is fully registered (appears in `UpdateAppList`)

### Summary
Suspend notifications to mobile until app HMI state is established

### Changelog
##### Bug Fixes
* Suspends notifications to mobile until app HMI state is established

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d [... cut]
closed 2021-11-09 by iCollin