smartdevicelink/sdl_core #3676: Incorrect behavior during processing of erroneous Slider responsive with `sliderPosition` parameter

GetmanetsIrina, 2021-03-25:
### Bug Report
Incorrect behavior during processing of erroneous Slider responsive with `sliderPosition` parameter

##### Precondition
1. HMI and SDL are started
2. Mobile app is registered and activated

##### Reproduction Steps
1. Mobile app requests Slider with default params and values
2. Press `Back`  button or wait until a timeout is expired

##### Expected Behavior
Mobile API says
```
<param name="sliderPosition" type="Integer" minvalue="1" maxvalue="26" mandatory="false">
    <description>
        Current slider value returned when saved or canceled (aborted)
        This value is only returned for resultCodes "SAVED" or "ABORTED"
    </description>
</param>
```

 HMI API says
```
<param name="sliderPosition" type="Integer" minvalue="1" maxvalue="26" mandatory="false">
  <description>Current slider position. Must be returned when the user has clicked the ‘Save’ or ‘Canceled’ button or by the timeout </description>
</param>
```

According to the information above HMI should send `sliderPosition` parameter also in erroneous response, but according to HMI specification, HMI can not provide custom parameters in the erroneous response. All faced custom parameters will be treated as fake parameters and will be removed from the message.
So we were thinking that the summary in API's should be updated and should say that `sliderPosition` parameter should be provided only in a successful response.
SDL should not provide `sliderPosition` parameter in erroneous response.

Note: 
HMI issue should be created to remove `sliderPosition` parameter from the erroneous responses in case this solution will be applied.

##### Observed Behavior
1. HMI sends `sliderPosition` parameter with actual value in erroneous responses(ABORTED and TIMED_OUT)
2. SDL sends `sliderPosition=0` in the response to the mobile side because can not process the `sliderPosition` received in erroneous response from HMI

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/7.1.0-RC1 (0df66080512cbe76ee5d4872910a54adf3ca1f36)
* Testing Against: HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop (7fad546)
SPT: 20210305-Android (1076) 

##### Logs
[slider_sdl_log.zip](https://github.com/smartdevicelink/sdl_core/files/6204024/slider_sdl_log.zip)

jacobkeeler, 2021-05-10:
@GetmanetsIrina It seems to me that this is at least partially an HMI issue as well. The HMI can send an ABORTED or TIMED_OUT response with the `sliderPosition` parameter, but it needs to use the normal response format rather than the error one.
```
       {
          'jsonrpc': '2.0',
          'id': sliderRequestID,
          'result': {
            'code': 5,
            'method': 'UI.Slider',
            'sliderPosition': 8
          }
        }
``` 
Unfortunately when using this format, the HMI is not able send a `message` field, but sending `sliderPosition` is possible.

The main issue on Core's end is that it sends a `sliderPosition` of `0` if the HMI doesn't send this value properly, which is not valid in the Mobile API since `minvalue="1"`. I don't believe that the Mobile API description needs to be changed in this case.

jacobkeeler, 2021-05-14:
Closing with the merge of #3697

PR #3697 (2021-05-14): Remove logic sending invalid slider position
Fixes #3676 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Manual testing against SDL HMI PR (https://github.com/smartdevicelink/sdl_hmi/pull/564), app should receive sliderPosition in ABORTED or TIMED_OUT response for Slider if provided by the HMI.

### Summary
sliderPosition was being set to 0 (an invalid value) in the mobile response for Slider if an ABORTED or TIMED_OUT result was received from the HMI without this parameter. This PR removes that logic.

### Changelog
##### Bug Fixes
* Fix invali [... cut]
closed 2021-05-14 by jacobkeeler