smartdevicelink/sdl_core #3683: SDL sends SubscribeWayPoints to HMI for 1st app even if there is a subscription for 2nd app.

dboltovskyi, 2021-04-08:
### Bug Report
SDL sends `SubscribeWayPoints` to HMI for 1st app even if there is a subscription for 2nd app.

##### Reproduction Steps
1. AddSubMenu related to resumption is sent by App1
2. App1 and App2 are subscribed to WayPoints
3. Unexpected disconnect and reconnect are performed
4. App1 and App2 re-register with actual HashId
=> SDL does:
   - start resumption process for App1 and App2
   - send UI.AddSubMenu and Navi.SubscribeWayPoints requests related to App1 to HMI
5. HMI responds with <erroneous> resultCode to UI.AddSubMenu and <successful> to Navi.SubscribeWayPoints
=> SDL does:
   - not send revert Navi.UnsbscribeWayPoints request to HMI
   - not restore subscription to WP for App1 and responds RAI_Response(success=true,resultCode=RESUME_FAILED) to App1
   - restore subscription to WP for App2 and responds RAI_Response(success=true,resultCode=SUCCESS) to App2
6. App1 tries to subscribe to WayPoints

##### Expected Behavior
SDL does not send `SubscribeWayPoints request` to HMI since there is an active subscription for another app

##### Observed Behavior
SDL does send `SubscribeWayPoints` request to HMI

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/7.1.0-RC1 (c163c2d)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script:
 - Report and Logs:

dboltovskyi, 2021-04-08:
@jordynmackool @JackLivio Please notice this is a regression issue. It became reproducible within https://github.com/smartdevicelink/sdl_core/pull/3573 and not reproduced on a previous commit 26ca4e31a3 We apologize for the late catch. ATF script attached is part of our new extended test coverage which has been created recently.

GetmanetsIrina, 2021-04-09:
The issue has been verified on [fix branch](https://github.com/smartdevicelink/sdl_core/tree/fix/issue_3683) (c9d46d1).
Verification status is Passed.
Regression status is also Passed on the fix branch.

PR #3684 (2021-04-09): Check waypoint subscription size in resumption
Fixes #3683 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
See reproduction steps in issue #3683 

### Summary
Check size of waypoint subscription list before unsubscribing HMI from way points.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2021-04-09 by JackLivio