smartdevicelink/sdl_core #3685: SDL starts PTU sequence by Days trigger on second ignition cycle

VjKlepikov, 2021-04-14:
### Bug Report
SDL starts PTU sequence by Days trigger on second ignition cycle

##### Reproduction Steps
0. "exchange_after_x_days": 30 in sdl_preloaded_pt.json
1. Start HMI and Core
2. Shut down Core and HMI via IGNITION_OFF signal
    Start HMI and Core again

##### Expected Behavior
1. SDL does not start PTU sequence

##### Observed Behavior
1. The PTU sequence is started by Days trigger before the previous UpdatedPollicyTable is successfully applied. 

/policy/policy_regular/src/policy_manager_impl.cc:1266 PolicyManagerImpl::ExceededDays: Enter
TRACE [02 Apr 2021 15:18:59,545][140210323048192][Policy] /home/slava/Work/SmartDeviceLink/sdl_core/src/components/policy/policy_regular/src/cache_manager.cc:661 CacheManager::DaysBeforeExchange: Enter
DEBUG [02 Apr 2021 15:18:59,545][140210323048192][Policy] /home/slava/Work/SmartDeviceLink/sdl_core/src/components/policy/policy_regular/src/cache_manager.cc:673 CacheManager::DaysBeforeExchange: Exchange after: 30 days
DEBUG [02 Apr 2021 15:18:59,545][140210323048192][Policy] /home/slava/Work/SmartDeviceLink/sdl_core/src/components/policy/policy_regular/src/cache_manager.cc:675 CacheManager::DaysBeforeExchange: Epoch since last update: 0
TRACE [02 Apr 2021 15:18:59,545][140210967283456][JsoncppReaderWrapper] /home/slava/Work/SmartDeviceLink/sdl_core/src/components/utils/src/jsoncpp_reader_wrapper.cc:45 JsonReader::parse: Enter
DEBUG [02 Apr 2021 15:18:59,545][140210323048192][Policy] /home/slava/Work/SmartDeviceLink/sdl_core/src/components/policy/policy_regular/src/cache_manager.cc:679 CacheManager::DaysBeforeExchange: The days since last update: 18719
TRACE [02 Apr 2021 15:18:59,545][140210323048192][Policy] /home/slava/Work/SmartDeviceLink/sdl_core/src/components/policy/policy_regular/src/cache_manager.cc:661 CacheManager::DaysBeforeExchange: Exit
TRACE [02 Apr 2021 15:18:59,545][140210323048192][Policy] /home/slava/Work/SmartDeviceLink/sdl_core/src/components/policy/policy_regular/src/policy_manager_impl.cc:1266 PolicyManagerImpl::ExceededDays: Exit
DEBUG [02 Apr 2021 15:18:59,545][140210323048192][Policy] /home/slava/Work/SmartDeviceLink/sdl_core/src/components/policy/policy_regular/src/policy_manager_impl.cc:136 PolicyManagerImpl::CheckTriggers: 
Days exceeded: true
Ignition cycles exceeded: false

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/7.1.0-RC1 (0df6608)

VjKlepikov, 2021-04-14:
Please notice issue is reproduced on release_7.0.0

jacobkeeler, 2021-05-14:
@VjKlepikov Could you clarify what is wrong with the current behavior? My understanding was that this trigger should fire if Core has never done a Policy Table exchange before. Generally it's expected that on a clean install, Core will attempt to perform a PTU on startup until at least one has been performed

dboltovskyi, 2021-05-18:
@jacobkeeler It looks like this is not a "bug", but a "feature".
SDL may start PTU sequence on a very first start (due to Days trigger). But this change may have a huge impact on existing ATF scripts.
Some scripts do not expect PTU start and would fail in that case.

I propose to leave SDL logic as is and to create a new ATF script to cover this specific sequence as a normal behavior.

jacobkeeler, 2021-05-20:
@dboltovskyi are there any tests that are failing due to this behavior at the moment?

dboltovskyi, 2021-06-09:
@jacobkeeler There are no such tests as of now

closed 2021-06-10 by jacobkeeler