smartdevicelink/sdl_core #3717: SDL doesn't ignore value of 'frameInfo' field for 'Single' and 'First' frames

dboltovskyi, 2021-06-09:
### Bug Report
SDL doesn't ignore value of 'frameInfo' field for 'Single' and 'First' frames.
It has to do so according to current protocol specification: [2.3 Frame Header Fields](https://github.com/smartdevicelink/protocol_spec#23-frame-header-fields) ("Frame Info")

##### Reproduction Steps
1. App registered
2. App sends PutFile data
  a) as a single frame
  b) as a multi frame
3. Make sure App sends non-zero value in 'frameInfo' field of 'Single' and 'First' frames

##### Expected Behavior
SDL does:
  - ignore value of 'frameInfo' field for 'Single' and 'First' frames
  - proceed with PutFile successfully

##### Observed Behavior
SDL fails to proceed frames with message in log:
```
IncomingDataHandler::CreateFrame: Packet validation failed
```

##### OS & Version Information
* OS/Version: Ubuntu 20.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (5227a8451e)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
 - ATF Script: 
 - Report and Logs:

dboltovskyi, 2021-06-10:
Issue is raised based on comment: https://github.com/smartdevicelink/sdl_evolution/issues/1070#issuecomment-856132544
Luxoft team is going to provide a fix.

jordynmackool, 2021-08-20:
Hi @dboltovskyi, Livio is going to begin Release candidate prep, starting on 2021-08-30 for the upcoming Core 8.0 release. As noted in your comment above, Luxoft is planning on contributing a PR for this issue. Can you please provide timing on when Luxoft will have this PR ready for Livio review? Thanks!

dboltovskyi, 2021-08-20:
@jordynmackool Please notice both PRs related to this issue are ready for PM review.

iCollin, 2021-08-24:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3762

PR #3762 (2021-08-24): Ignore frame data for single and first frames
Fixes #3717 

This PR is **ready** for review.

### Risk
This PR makes **no]** API changes.

### Testing Plan
1. App registered
2. App sends PutFile data
  a) as a single frame
  b) as a multi frame
3. Make sure App sends non-zero value in 'frameInfo' field of 'Single' and 'First' frames

### Summary
Updated validation for single and first frame

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2021-08-24 by iCollin