smartdevicelink/sdl_core #3722: [SDL 0338] Remove Unused Files From SDL Core

theresalech, 2021-06-16:
Proposal: [Remove Unused Files From SDL Core](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0338-remove-unused-files-from-sdl-core.md)

> This proposal is to remove some unused utility and RPC files in SDL Core.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/1152

ychernysheva, 2021-06-16:
Hi @JackLivio 
We have noticed that there are another unused notifications in SDLRPCPlugin code base, besides of OnVIBeltStatusNotification and OnVIAccPedalPositionNotification. It seems that
OnVIBodyInformationNotification,
OnVIDeviceStatusNotification,
OnVIDriverBrakingNotification,
OnVIEngineTorqueNotification,
OnVIExternalTemperatureNotification,
OnVIFuelLevelNotification,
OnVIFuelLevelStateNotification, 
OnVIGpsDataNotification,
OnVIHeadLampStatusNotification, 
OnVIInstantFuelConsumptionNotification,
OnVIMyKeyNotification, 
OnVIOdometerNotification,
OnVIPrndlNotification,
OnVIRpmNotification,
OnVISpeedNotification, 
OnVISteeringWheelAngleNotification,
OnVITirePressureNotification,
OnVIVinNotification,
OnVIWiperStatusNotification 
are also unused and probably should to be removed. May I kindly suggest to include them in the scope of this proposal, or my team may create separate issue for this task?

JackLivio, 2021-06-17:
Thank you for the note, I will look at revising the accepted proposal.

theresalech, 2021-06-30:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0338 Revisions - Remove Unused Files From SDL Core](https://github.com/smartdevicelink/sdl_evolution/issues/1157).  Revisions include additional unused files to be removed.  Please reference [this pull request](https://github.com/smartdevicelink/sdl_evolution/pull/1156) for details.

theresalech, 2021-08-25:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0338 Revisions - Remove Unused Files From SDL Core](https://github.com/smartdevicelink/sdl_evolution/issues/1175).  Revisions include keeping two files that were found to be used in the project. Please reference [this pull request](https://github.com/smartdevicelink/sdl_evolution/pull/1173) for details.