smartdevicelink/sdl_core #3780: SDL doesn't stop after IGNITION_OFF in case mobile connected via USB

VNikiforov-dev, 2021-09-27:
### Bug Report
SDL doesn't stop after IGNITION_OFF in case mobile connected via USB

##### Preconditions
1. SDL and HMI are started
2. Connect mobile device via USB transport
##### Reproduction Steps
1. Perform ignition Off:
On "HMI" go to → Click the "Exit Application" button → Select “IGNITION_OFF” in “Exit Application reason” → Click the `Send state` button
SDL->HMI: BasicCommunication.OnExitAllApplications:{"reason":"IGNITION_OFF"}}

##### Expected Behavior
SDL stopped

##### Observed Behavior
SDL doesn't stopped

##### OS & Version Information
* OS/Version: [Ubuntu 18]
* SDL Core Version: [ https://github.com/smartdevicelink/sdl_core/tree/release/8.0.0 (5da5770 )]
* HMI: [https://github.com/smartdevicelink/sdl_hmi/tree/release/5.6.0 (ceb8b3c)]
* SPT: 20210729-Android

VNikiforov-dev, 2021-09-27:
Please note Luxoft team is going to prepare a fix

dboltovskyi, 2021-10-12:
@jordynmackool Please notice this is a regression issue. Fix [pull/3785](https://github.com/smartdevicelink/sdl_core/pull/3785) is ready for Livio team review. We would like it to be included in `8.0` release. Thank you.