smartdevicelink/sdl_core #3783: Core dump during language change

GetmanetsIrina, 2021-10-05:
### Bug Report
Core dump during language change for several apps

##### Reproduction Steps
1. SDL and HMI are started
2. 2-10 apps are registered
3. apps are activated one by one
4. Go to Apps menu
5. Change VR language to ES-MX (maybe update ui HMI language to some another value before )

##### Expected Behavior
All apps are reregistered successfully 

##### Observed Behavior
core dump

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/releases/tag/8.0.0-RC1
* HMI: https://github.com/smartdevicelink/sdl_hmi/tree/release/5.6.0 (1fb8b6b)
* SPT: 20210729-Android

[logs_dump.zip](https://github.com/smartdevicelink/sdl_core/files/7285648/logs_dump.zip)
[logs_dump_2.zip](https://github.com/smartdevicelink/sdl_core/files/7285652/logs_dump_2.zip)

dboltovskyi, 2021-10-12:
@theresalech Please notice fix for this issue is under development by Luxoft team. We are planning to submit a PR tomorrow.

iCollin, 2021-10-14:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3796

closed 2021-10-14 by iCollin