smartdevicelink/sdl_core #3817: Core doesn't register app in some scenarios during resumption

dboltovskyi, 2021-11-05:
### Bug Report
Core doesn't register app in some scenarios during resumption

##### Occurrences
Rare 

##### Reproduction Steps
1. There are 2 apps registered from the same device
2. Both apps added some resumption data
3. Unexpected disconnect happened
4. Device re-connected and both apps started re-register
5. HMI provides an error response for some of the resumption data requests

##### Expected Behavior
Both apps re-register successfully

##### Observed Behavior
One of the apps doesn't receive RAI response

##### OS & Version Information
* OS/Version: Ubuntu 20.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (31324d5)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
- Impacted scripts:
    - https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Resumption/Handling_errors_from_HMI/007_resume_all_data_with_one_error_rpc_2apps_unexpected_disconnect.lua
    - https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Resumption/Handling_errors_from_HMI/008_resume_all_data_with_one_error_rpc_2apps_ign_off.lua
- Report and Logs:

jordynmackool, 2021-11-08:
@dboltovskyi please advise if Luxoft plans to submit an issue for this PR?

dboltovskyi, 2021-11-08:
@jordynmackool Yes, Luxoft team is going to provide a fix.

closed 2022-01-25 by iCollin