smartdevicelink/sdl_core #3821: Core doesn't restore app's HMI level during resumption

dboltovskyi, 2021-11-08:
### Bug Report
Core doesn't restore app's HMI level during resumption

##### Occurrences
Rare

##### Reproduction Steps
1. App registered and activated (HMI level is `FULL`)
2. App adds `Command1`, `Command2`, `Command3` commands with `vrCommands` parameter
3. Perform disconnect and reconnect

##### Expected Behavior
SDL restore app's HMI level to `FULL`

##### Observed Behavior
SDL doesn't restore app's HMI level

##### OS & Version Information
* OS/Version: Ubuntu 20.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (31324d5)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
- Impacted scripts: 
    - https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/SDL5_0/Handling_VR_help_requests/023_SetGlobalProp_custom_vrHelp_only_by_resumption.lua 
- Reports and logs:

dboltovskyi, 2021-11-08:
@theresalech Please notice Luxoft team is planning to provide a fix

iCollin, 2022-01-06:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3836

PR #3836 (2022-01-06): Fix restoring HMI application level
Fixes #3821 

This PR is **[ready ]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
ATF

### Summary
The reason why HMI level was not restored is because the `State Controller` was not subscribed on `ActivateApp` event.
There is noticed the `State Controller` subscribes in runtime on events `CloseApplication` and `ActivateApp` and then never unsubscribes from it. In PR the subscription condition has been changed to prevent situation when we mistakenly did not subscribe on event. And also added unsubscribe for these events.

###  [... cut]
closed 2022-01-06 by iCollin