smartdevicelink/sdl_core #3823: Core closes web socket connection with 'unspecified system error' during RAI for WSS

dboltovskyi, 2021-11-08:
### Bug Report
Core closes web socket connection with 'unspecified system error' during RAI for WSS

##### Occurrences
Rare

##### Reproduction Steps
1. App registers using `WSS` connection

##### Expected Behavior
Core does register App successfully

##### Observed Behavior
Core closes web socket connection with error:
```
ERROR [05 Oct 2021 23:40:39,880][140250370254592][WebSocketSession] /home/jenkins/workspace/Release_8.0_SDL_NoUT_E/src/components/transport_manager/src/websocket_server/websocket_session.cc:127 transport_adapter::WebSocketSession<ExecutorType>::Read: unspecified system error
```

##### OS & Version Information
* OS/Version: Ubuntu 20.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (31324d5)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
- Scripts: `<any>`
- Reports and logs:
    -

dboltovskyi, 2021-11-08:
@theresalech Please notice Luxoft team is planning to provide a fix

iCollin, 2022-02-03:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3851

closed 2022-02-03 by iCollin