smartdevicelink/sdl_core #3824: Core crash during streaming switching

dboltovskyi, 2021-11-08:
### Bug Report
Core crash during streaming switching

##### Reproduction Steps
1. Set `StopStreamingTimeout` = 3000 and `VideoDataStoppedTimeout` = 3000 in SDL .INI file
2. Start SDL, HMI, connect Mobile device
3. Register 2 NAVIGATION applications: `App_1` and `App_2`
4. Activate `App_1` and start Video streaming
5. Deactivate `App_1` (streaming still continue since app has `STREAMABLE` state)
6. Activate `App_2` and start Video streaming
7. `App_1` and `App_2` continue streaming data within `StopStreamingTimeout` timeout
8. `App_1` stops streaming within `StopStreamingTimeout` timeout

##### Expected Behavior
Core does:
    - switch streaming between apps and provides HMI with streaming data from App_2
    - sends `Navi.StopStream` for `App_1`
    - sends `Navi.OnVideoDataStreaming(false)`
    - sends `Navi.StartStream` for `App_2`
    - sends `Navi.OnVideoDataStreaming(true)`
    - not unregister `App_1` since timeout is not yet expired

##### Observed Behavior
Core crash

##### OS & Version Information
* OS/Version: Ubuntu 20.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (31324d5)
* Testing Against: ATF

##### Test Case, Sample Code, and / or Example App
- Impacted scripts: 
    - https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/6_1/3142_01_Absence_of_misplaced_OnVideoDataStreaming.lua
    - https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/6_1/3142_02_Absence_of_misplaced_OnAudioDataStreaming.lua 
- Reports and logs:
    -

dboltovskyi, 2021-11-08:
@theresalech Please notice Luxoft team is planning to provide a fix

iCollin, 2021-12-22:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3833

closed 2021-12-22 by iCollin