smartdevicelink/sdl_core #3829: SDL does not resend OnButton notifications for custom buttons that were added within ShowConstantTBT, UpdateTurnList RPCs

VjKlepikov, 2021-11-19:
### Bug Report
SDL does not resend OnButton notifications for custom buttons that were added within ShowConstantTBT, UpdateTurnList RPCs

##### Reproduction Steps
1. SDL and HMI are started
2. Mobile app has assigned "Navigation-1" policy group
3. Mobile app is registered and activated (SDL automatically adds subscription for the CUSTOM_BUTTONS during registration)
4. Mobile app requests ShowConstantTBT and UpdateTurnList RPCs with softButtons(DEFAULT_ACTION)
5. Press added soft buttons
HMI sends OnButtonPress and OnButtonEvent notifications to SDL

##### Expected Behavior
SDL resends OnButtonPress and OnButtonEvent notifications to the mobile app

##### Observed Behavior
SDL does not resend  OnButtonPress and OnButtonEvent notifications to the mobile app

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (1e88e6f3)
* HMI: https://github.com/smartdevicelink/sdl_hmi/tree/develop

##### Test Case, Sample Code, and / or Example App
[Paste a link to a PR, gist, or other code that exemplifies this behavior]

VjKlepikov, 2021-11-19:
@theresalech Please notice Luxoft team is planning to provide a fix and test scripts

iCollin, 2022-01-11:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3839

closed 2022-01-11 by iCollin