smartdevicelink/sdl_core #3831: Core doesn't utilize samplingRate и bitsPerSample parameters provided by HMI when guessing playback time

dboltovskyi, 2021-11-23:
### Bug Report
Within fix [pull/3057](https://github.com/smartdevicelink/sdl_core/pull/3057) Core tries to guess playback time using formula:
```
latency_compensation = 500;
timeout = 1000 * data_size / (sampling_rate_ * bits_per_sample_ / 8) + latency_compensation;
```
where `sampling_rate_` and `bits_per_sample_` are the values provided by HMI within `pcmStreamCapabilities`

However Core always use hardcoded values for `bitsPerSample=16` and `samplingRate=16000` and doesn't take into account the ones provided by HMI.

##### Reproduction Steps
1. Make sure HMI provides non-default `bitsPerSample` and `samplingRate` values (e.g. `samplingRate=32000`)
2. Set `AudioStreamConsumer` to `socket` and `AudioStreamingPort` to `5080` in `smartDeviceLink.ini`.
3. Start audio stream from app (accept prompt from HMI to start audio stream)
4. Start reading audio stream using the following `gstreamer` command:
```
time gst-launch-1.0 souphttpsrc location=http://127.0.0.1:5080 ! audio/x-raw,format=S16LE,rate=32000,channels=1 ! pulsesink
```
5. Notice time when `gstreamer` exits (socket is closed)

##### Expected Behavior
Socket is closed once audio stopped playing

##### Observed Behavior
Socket remains open some time after audio stopped playing.
Looks like the values from `pcmStreamCapabilities` provided by HMI wasn't taken into account.

##### OS & Version Information
* OS/Version: Ubuntu 20.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/develop (31324d5)
* Testing Against: SPT

dboltovskyi, 2021-11-23:
@theresalech Please notice Luxoft team is going to provide a fixf for this issue

jacobkeeler, 2022-01-11:
Closed via #3837

closed 2022-01-11 by jacobkeeler