smartdevicelink/sdl_core #3838: SDL sends redundant UI.ClosePopUp request in case PerformInteraction VR only

VjKlepikov, 2021-12-17:
### Bug Report
SDL sends redundant UI.ClosePopUp request in case PerformInteraction VR only

##### Reproduction Steps
1. SDL and HMI are started
2. Mobile app is registered and activated
3. CreateInteractionChoiceSet is added
4.  App sends PerformInteraction request with VR_ONLY interaction mode to SDL
5. HMI responds with 'SUCCESS' to UI.PerformInteraction request
6. HMI responds with 'TIMED_OUT' to VR.PerformInteraction request

##### Expected Behavior
SDL does not send UI.ClosePopUp request to HMI 

##### Observed Behavior
SDL sends UI.ClosePopUp request to HMI

##### OS & Version Information
OS/Version: Ubuntu 18.04
* SDL Core Version: [https://github.com/smartdevicelink/sdl_core/tree/develop](https://github.com/smartdevicelink/sdl_core/tree/develop)

VjKlepikov, 2021-12-17:
@theresalech Please notice Luxoft team is going to prepare a fix and ATF script(s) to verify it.

jacobkeeler, 2022-02-11:
closed via #3844

PR #3844 (2022-02-11): Fix perform interaction close pop up
Fixes #3838 

This PR is **[ready ]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
ATF and manual

### Summary
In the PR we prevent to sending redundant UI.ClosePopUp request for VR interaction mode

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2022-02-11 by jacobkeeler