smartdevicelink/sdl_core #3840: sdl_core: missing 'Create' symbol in libPolicy.so

WinerSun, 2022-01-04:
### Bug Report
[Summary]
sdl_core/src/components/application_manager/src/plugin_manager/rpc_plugin_manager_impl.cc:41 plugin_manager::GetFuncFromLib: Failed to export symbols : .//libPolicy.so: undefined symbol: Create

##### Reproduction Steps
run `sdl_core`, then occured this error and the sdl_core can not running

##### Expected Behavior
run `sdl_core`, then sdl_core process can run normally.

##### Observed Behavior

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: commit 68f082169e0a40fccd9eb0db3c83911c28870f07

JackLivio, 2022-01-25:
@WinerSun If using SDL Core 8.0.0, please upgrade your Ubuntu to version 18 or 20. We dropped support for Ubuntu 16 in the SDL Core 8.0.0 release. 

Previous versions of SDL Core will work on Ubuntu 16. Please let me know if you are experiencing issues after changing your version of Core or Ubuntu.

JackLivio, 2022-03-17:
Closed for inactivity :) feel free to reopen issue

closed 2022-03-17 by JackLivio