smartdevicelink/sdl_core #3846: SDL removes the current "consumer_friendly_messages" section in Local PT in case the Updated PT omits "consumer_friendly_messages" section

VjKlepikov, 2022-01-21:
### Bug Report
SDL removes the current "consumer_friendly_messages" section in Local PT in case the Updated PT omits "consumer_friendly_messages" section

##### Reproduction Steps
1. Make sure SDL is built with HTTP flow
2. Register new application => PTU is triggered
3. PTU omits "consumer_friendly_messages" section

##### Expected Behavior
SDL maintains the current "consumer_friendly_messages" section in Local PT

##### Observed Behavior
SDL removes the current "consumer_friendly_messages" section in Local PT in case the Updated PT omits "consumer_friendly_messages" section

##### OS & Version Information
* OS/Version: Ubuntu 18
* SDL Core Version:  https://github.com/smartdevicelink/sdl_core/tree/develop (6ca6961)

##### Test Case, Sample Code, and / or Example App
* Tests scripts: *./test_scripts/Policies/build_options/070_ATF_PTU_Merge_Into_Local_PT_ConsumerFriendlyMessages_Omitted_HTTP.lua

#####  Logs: [Logs.zip](https://github.com/smartdevicelink/sdl_core/files/7914132/Logs.zip)

VjKlepikov, 2022-01-31:
Please notice this is a regression issue that became reproducible after merge of [#3842](https://github.com/smartdevicelink/sdl_core/pull/3842)
@theresalech  Can you please advise if the Livio team is planning on submitting a PR to fix?

theresalech, 2022-01-31:
@VjKlepikov our team will review the issue and advise.

jacobkeeler, 2022-01-31:
@VjKlepikov can you confirm if this issue only occurs in HTTP mode?

jacobkeeler, 2022-01-31:
After testing, it appears that this issue is reproducible in master if `OnDriverDistraction(DD_ON)` is sent prior to registering the app. It seems likely that the issue lies in [this section of code](https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/application_manager/src/application_manager_impl.cc#L4663-L4678)

jacobkeeler, 2022-02-04:
closed via #3852

PR #3852 (2022-02-04): Check that message code exists before accessing

Fixes #3846 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
- Run ATF test `test_scripts/Policies/build_options/070_ATF_PTU_Merge_Into_Local_PT_ConsumerFriendlyMessages_Omitted_HTTP.lua` with HTTP policies

### Summary
The `GetUserFriendlyMsg` function was inadvertently creating an empty messages entry when searching for a message code which didn't exist in the policy table during the PTU process, causing the table to be corrupted. This PR adds a check to this function to prevent this from happeni [... cut]
closed 2022-02-04 by jacobkeeler