smartdevicelink/sdl_core #3855: SDL crash during PTU

VjKlepikov, 2022-02-11:
### Bug Report
SDL crash during PTU

##### Reproduction Steps
1. SDL and HMI are started
2. Register App
3. PTU is triggered

##### Expected Behavior
Core does not crash

##### Observed Behavior
PTU finished successfully. No crash.

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: [2f00229](https://github.com/smartdevicelink/sdl_core/tree/develop)

##### Test Case, Sample Code, and / or Example App
./test_scripts/Policies/App_Permissions/005_ATF_DISALLOWED_app_Id_policies_And_RequestType_Validation.lua

##### 
Logs:[2022-02-11_12-35-16.685_failed.tar.gz](https://github.com/smartdevicelink/sdl_core/files/8048816/2022-02-11_12-35-16.685_failed.tar.gz)

OlhaVorobiova, 2022-02-15:
@theresalech Please notice Luxoft team is going to prepare a fix

VjKlepikov, 2022-02-18:
@theresalech Please notice this is a non-regression issue. It's also reproduced on 8.0

JackLivio, 2022-03-09:
Fixed via #3864

VjKlepikov, 2022-03-17:
Issue is reproduced on release/8.1.0 (cf9c386)
Core crash during PTU in ./test_scripts/RC/CLIMATE_RADIO/SetInteriorVehicleData/004_Disallow_flow_in_case_appHMIType_is_not_REMOTE_CONTROL.lua
logs: [2022-03-14_02-49-41.448_failed.tar.gz](https://github.com/smartdevicelink/sdl_core/files/8283339/2022-03-14_02-49-41.448_failed.tar.gz)
 @theresalech Please notice Luxoft team is going to prepare a fix