smartdevicelink/sdl_core #3857: SDL core dump on Ignition OFF during registering Apps

VjKlepikov, 2022-02-14:
## Bug Report
SDL core dump on Ignition OFF during registering Apps

##### Reproduction Steps
1. SDL and HMI are started
2. Connect mobile device via TCP transport.
3. Perform ignition Off during registering 20 Apps
On "HMI" go to → Click the "Exit Application" button → Select “IGNITION_OFF” in “Exit Application reason” → Click the `Send state` button

##### Expected Behavior
SDL starts the ignition off sequence
1. SDL sends to HMI OnSDLClose and stops working
SDL->HMI: BasicCommunication.OnSDLClose:{}

##### Observed Behavior
SDL starts the ignition off sequence
1. SDL sends to HMI OnSDLClose and stops working
SDL->HMI: BasicCommunication.OnSDLClose:{}
Core dump is observed
SIGSEGV caught
DEBUG [139969695552064][SDLMain] /home/jenkins/workspace/Develop_SDL_NoUT_P/src/appMain/life_cycle_impl.cc:241 {anonymous}::sig_handler: SIGSEGV signal has been caught

##### OS & Version Information
* OS/Version: Ubuntu 18
* SDL Core Version:  https://github.com/smartdevicelink/sdl_core/tree/develop (7a4a4e0)

##### Logs:
[Logs.zip](https://github.com/smartdevicelink/sdl_core/files/8061975/Logs.zip)

VjKlepikov, 2022-02-14:
Please notice this is a non-regression issue. It's also reproduced on 8.0

VjKlepikov, 2022-02-14:
@theresalech Please notice Luxoft team is going to prepare a fix

ShobhitAd, 2022-06-27:
@VjKlepikov I was able to reproduce this issue on my end manually but might need an ATF tests PR to more easily test the fix PR.

VjKlepikov, 2022-06-30:
@ShobhitAd Please find PR with  the script in https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2694

ShobhitAd, 2022-08-25:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3939

PR #3939 (2022-08-25): Core dump on ignition off when registering 20 apps
Fixes #3857 

Original PR: https://github.com/smartdevicelink/sdl_core/pull/3928

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Submitted ATF test for issue

### Summary
Fixes the following problems:

- Аn object of this class AppExtension was not given externally without protection. Therefore, I will change the return type from Extensions () to DataAccessor <std :: list >. Where the desired mutex is captured.
- ApplicationManagerImpl had unprotected deletion of objects used in other threads. Al [... cut]
closed 2022-08-25 by ShobhitAd