smartdevicelink/sdl_core #3858: SDL sends wrong CreateInteractionChoiceSet response  in case success type of response(resultCode: WARNINGS, RETRY, SAVED, WRONG_LANGUAGE, TRUNCATED_DATA) from HMI

VjKlepikov, 2022-02-15:
### Bug Report
SDL sends wrong CreateInteractionChoiceSet response  in case success type of response(resultCode: WARNINGS, RETRY, SAVED, WRONG_LANGUAGE, TRUNCATED_DATA) from HMI

##### Reproduction Steps
1. SDL and HMI are started
2. Mobile app is registered and activated
3. Mobile app requests CreateInteractionChoiceSet
4. HMI responds with 7 result code(RETRY) in result structure

##### Expected Behavior
Mobile app receives CreateInteractionChoiceSet response with resultCode=RETRY, success=true

##### Observed Behavior
Mobile app receives CreateInteractionChoiceSet response with resultCode=RETRY, success=false

Note: SDL send CreateInteractionChoiceSet response "success=true, resultCode=SUCCESS"  in case success type of response WARNINGS

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: [https://github.com/smartdevicelink/sdl_core/tree/develop (b74c738)](https://github.com/smartdevicelink/sdl_core/tree/develop)

##### Logs:
[Logs.zip](https://github.com/smartdevicelink/sdl_core/files/8068358/Logs.zip)

VjKlepikov, 2022-02-15:
Please notice this is a non-regression issue. It's also reproduced on 8.0

ShobhitAd, 2022-06-15:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3907

closed 2022-06-15 by ShobhitAd