smartdevicelink/sdl_core #3862: SDL rejects SystemRequest from mobile App during PTU

VjKlepikov, 2022-02-16:
### Bug Report
SDL rejects SystemRequest from mobile App during PTU

##### Reproduction Steps
1. Core and HMI are started
2. App1 and App2 are registered
3. Core sends OnStatusUpdate UPDATING
4. SDL send OnSystemRequest to APP1
5. App1 sends SystemRequest(HTTP) to SDL on OnSystemRequest notification

##### Expected Behavior
SDL sends SystemRequest(SUCCESS) response to mobile App

##### Observed Behavior
SDL rejects SystemRequest from mobile App1

##### OS & Version Information
* OS/Version: Ubuntu 18
SDL: develop (b74c738)
SCR: develop https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/develop ([c7e8aa2](https://github.com/smartdevicelink/sdl_atf_test_scripts/commit/c7e8aa2f94bd48f11a260cc501952e344ac2c90a))
ATF: develop

##### Test Case, Sample Code, and / or Example App
./test_scripts/Policies/build_options/079_ATF_PTU_HMI_Level_Affected_Apps_NONE_BACKGROUND_HTTP.lua
./test_scripts/TheSameApp/Policies/AppId/002_PTU_is_triggered_after_registration_app_with_different_app_id.lua

Logs:  
[Logs.zip](https://github.com/smartdevicelink/sdl_core/files/8078707/Logs.zip)

VjKlepikov, 2022-02-16:
Please notice this is a regression issue that became reproducible after merge of [#3853](https://github.com/smartdevicelink/sdl_core/pull/3853)

iCollin, 2022-02-16:
Hello @VjKlepikov,

Are you sure this is an issue with Core? From a quick look at both tests, I see the log
`PolicyHandler::IsPTUSystemRequestAllowed: PTU received from unexpected application, request was sent to 0` It seems to me that the tests will need additional updates to trigger Core to send OnSystemRequest to the application which will provide the PTU in SystemRequest. Per 0337 an app may not PTU unless Core has specifically requested an update with OnSystemRequest.

I will look at adding the newly required updates to these tests now. Please let me know if I have misunderstood.

Thank you!

iCollin, 2022-02-16:
@VjKlepikov I do see the Core issue now, my mistake. The `last_ptu_app_id_` is not being updated by OnSystemRequest notifications from the HMI in HTTP policy mode. I will create a fix to address this now.

iCollin, 2022-02-21:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3863

closed 2022-02-21 by iCollin