smartdevicelink/sdl_core #3871: Core crash during App unregistered with PROTOCOL_VIOLATION

VjKlepikov, 2022-03-01:
### Bug Report
Core crash during App unregistered with PROTOCOL_VIOLATION

##### Occurrences: ~0.1%

##### Reproduction Steps
1. Set StopStreamingTimeout = 3000 in SDL .INI file
2. Start SDL, HMI, connect Mobile device
3. Register PROJECTION (App_1) and NAVIGATION (App_2) applications
4. Activate App_1 and start Video streaming
5. Deactivate App_1 (streaming still continue since app has STREAMABLE state)
6. Activate App_2

##### Expected Behavior
SDL does:
send OnHMIStatus(FULL, STREAMABLE) notification to App_2
  send OnHMIStatus(LIMITED, NOT_STREAMABLE) notification to App_1
  send Navi.OnVideoDataStreaming(false) notification to HMI
  start 1st 'StopStreamingTimeout' timeout
  
 Once timeout is expired SDL does:
     - send Navi.StopStream(App_1) request to HMI
     - send EndService(VIDEO) request to App_1
     - start 2nd 'StopStreamingTimeout' timeout

##### Observed Behavior
Core crash

##### OS & Version Information
OS/Version: Ubuntu 18.04
SDL Core Version: 8.0 + develop (8.1 release candidate)

##### Test Case, Sample Code, and / or Example App
./test_scripts/Defects/6_1/3139_03_Stop_streaming_forcibly_app_continue_streaming.lua

##### Logs:
[2022-02-28_05-18-50.845_failed.tar.gz](https://github.com/smartdevicelink/sdl_core/files/8164132/2022-02-28_05-18-50.845_failed.tar.gz)

VjKlepikov, 2022-03-01:
Please notice this is a non-regression issue. It's also reproduced on release 8.0

ShobhitAd, 2022-06-29:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3920

PR #3920 (2022-06-29): Make shared pointer atomic
Fixes #3871

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Atf scripts

### Summary
This is actually a template specialization proposed here:
https://en.cppreference.com/w/cpp/memory/shared_ptr/atomic2

If multiple threads of execution access the same std::shared_ptr object without synchronization and any of those accesses uses a non-const member function of shared_ptr then a data race will occur unless all such access is performed through an instance of std::atomic<<std::shared_ptr>>

To avoid these data races, shared [... cut]