smartdevicelink/sdl_core #3881: SDL rejects SystemRequest from mobile App during PTU in case HMI sends BC.OnSystemRequest with 'url' parameter

VjKlepikov, 2022-03-22:
### Bug Report
SDL rejects SystemRequest from mobile App during PTU in case HMI sends BC.OnSystemRequest with url parameter

##### Reproduction Steps
SDL is built in PROPRIETARY policy mode
Core and HMI are started
App is registered
SDL sends OnStatusUpdate(UPDATING)
HMI sends OnSystemRequest with 'url' parameter
SDL resend OnSystemRequest to App
App sends SystemRequest(PROPRIETARY) to SDL 

##### Expected Behavior
SDL sends SystemRequest(SUCCESS) response to mobile App

##### Observed Behavior
SDL rejects SystemRequest from mobile App

##### OS & Version Information
* OS/Version: Ubuntu 18
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/8.1.0 (25e47c7)

##### Logs:
[Logs_P.zip](https://github.com/smartdevicelink/sdl_core/files/8323477/Logs_P.zip)

VjKlepikov, 2022-03-22:
@theresalech  Please notice this is a regression issue that became reproducible after merge of https://github.com/smartdevicelink/sdl_core/pull/3853

JackLivio, 2022-03-22:
@VjKlepikov Can you provide the test you were using for this issue please?

VjKlepikov, 2022-03-22:
@JackLivio Issue has been found during manual testing with enable External Policies option  on Web-HMI 
Also please find draft script to check the issue https://github.com/smartdevicelink/sdl_atf_test_scripts/commit/c87e1a0fa1abadd46573b046d523e572d58923bd

JackLivio, 2022-03-23:
@VjKlepikov I confirmed my pr fixes the issue for both external and proprietary policy builds. Could you please open a pr with those atf changes?

VjKlepikov, 2022-03-24:
@JackLivio Yes, Luxoft team is going to prepare ATF script(s) to verify it

VjKlepikov, 2022-03-24:
@JackLivio Please find PR with ATF script(s) to check the issue
Please note SDL rejects SystemRequest from mobile App during PTU for several apps in case HMI sends BC.OnSystemRequest with 'url' parameter
Sometimes test script is FAILED 
https://github.com/LuxoftSDL/sdl_atf_test_scripts/blob/defects/3881/test_scripts/Policies/Policy_Table_Update/177_PTU_is_performed_for_several_Apps_if_HMI_sends_BC_OnSystemRequest_with_url_parameter.lua
logs: 
[Logs.zip](https://github.com/smartdevicelink/sdl_core/files/8342330/Logs.zip)
SDL: fix/issue_3881 (bf9e4e4d)

PR #3883 (2022-04-06): Obtain appID if url is sent to prop policies
Fixes #3881

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF policy test scripts.

### Summary
Normally in proprietary policies a url is not included in the system request. In the policy handler `GetNextUpdateUrl()` will find an app id to send the ptu to if an app id is not provided. This method is not called when a url is provided by the hmi. A call to ChoosePTUApplication() was added to handle the missing app id.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJ [... cut]
closed 2022-04-06 by JackLivio