smartdevicelink/sdl_core #3882: SDL does not send UI.ChangeRegistration(AppHMIType) request to HMI during PTU

VjKlepikov, 2022-03-22:
### Bug Report
SDL does not send UI.ChangeRegistration(AppHMIType) to HMI in case the app's AppHMIType received during Policies update with different from app's AppHMIType requested during registration

##### Reproduction Steps
1. SDL, HMI are started
2. App is registered with 'MEDIA' appHMIType
3. PTU is performed with 'REMOTE_CONTROL' appHMIType for App

##### Expected Behavior
SDL sends UI.ChangeRegistration(appID, AppHMIType) request to HMI 

##### Observed Behavior
SDL does not send UI.ChangeRegistration(appID, AppHMIType) request to HMI

##### OS & Version Information
OS/Version: Ubuntu 18
SDL Core Version: develop,  https://github.com/smartdevicelink/sdl_core/tree/release/8.1.0 (25e47c7)

##### Logs:
[Logs.zip](https://github.com/smartdevicelink/sdl_core/files/8325169/Logs.zip)

LitvinenkoIra, 2022-06-10:
Luxoft team is going to fix this issue

jacobkeeler, 2022-06-21:
Closed via #3927