smartdevicelink/sdl_core #3887: smartdevicelink project security check

alelaij, 2022-03-23:
### Bug Report

1. jsoncpp throw exception which caused the program to terminal.
  
2. byte_array_to_bson_string signed - unsigned conversion error
  
3. docString buffer overflow
  

##### Reproduction Steps

SDL and HMI are started  
App is registered

Send payload from tcp port

##### Expected Behavior

SDL security check

##### Observed Behavior

progress terminal

##### OS & Version Information

- Ubuntu Linux 20.04 with GCC 9.3.x
- SDL Core Version: 8.0 + develop (8.1 release candidate)

##### Test Case, Sample Code, and / or Example App

case-1

crash info

```c
bt
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007f5a0ee08859 in __GI_abort () at abort.c:79
#2  0x00007f5a0f1dd911 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007f5a0f1e938c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007f5a0f1e93f7 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007f5a0f1e96a9 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x000055f14f1b5d9e in Json::throwLogicError (msg="in Json::Value::resolveReference(key, end): requires objectValue") at /home/alex/Downloads/sdl_core/src/3rd_party-static/jsoncpp/src/lib_json/json_value.cpp:211
#7  0x000055f14f1ba240 in Json::Value::resolveReference (this=0x7f59bdc28440, key=0x55f14f3e32f9 "id", end=0x55f14f3e32fb "") at /home/alex/Downloads/sdl_core/src/3rd_party-static/jsoncpp/src/lib_json/json_value.cpp:1050
#8  0x000055f14f1ba9b0 in Json::Value::operator[] (this=0x7f59bdc28440, key=0x55f14f3e32f9 "id") at /home/alex/Downloads/sdl_core/src/3rd_party-static/jsoncpp/src/lib_json/json_value.cpp:1107
#9  0x000055f14f102374 in security_manager::SecurityManagerImpl::ProcessInternalError (this=0x55f150baeae0, inMessage=...) at /home/alex/Downloads/sdl_core/src/components/security_manager/src/security_manager_impl.cc:614
#10 0x000055f14f0f7b55 in security_manager::SecurityManagerImpl::Handle (this=0x55f150baeae0, message=...) at /home/alex/Downloads/sdl_core/src/components/security_manager/src/security_manager_impl.cc:137
#11 0x000055f14f109ce4 in threads::MessageLoopThread<utils::PrioritizedQueue<security_manager::SecurityMessage> >::LoopThreadDelegate::DrainQue (this=0x55f1505ae170) at /home/alex/Downloads/sdl_core/src/components/include/utils/threads/message_loop_thread.h:201
#12 0x000055f14f109b07 in threads::MessageLoopThread<utils::PrioritizedQueue<security_manager::SecurityMessage> >::LoopThreadDelegate::threadMain (this=0x55f1505ae170) at /home/alex/Downloads/sdl_core/src/components/include/utils/threads/message_loop_thread.h:184
#13 0x000055f14f194d1f in threads::Thread::<lambda(threads::Thread*)>::operator()(threads::Thread *) const (__closure=0x7f59bdc28b61, thread=0x55f1505b75f0) at /home/alex/Downloads/sdl_core/src/components/utils/src/threads/thread_posix.cc:76
#14 0x000055f14f195478 in threads::Thread::threadFunc (arg=0x55f1505b75f0) at /home/alex/Downloads/sdl_core/src/components/utils/src/threads/thread_posix.cc:106
#15 0x00007f5a0fb2a609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#16 0x00007f5a0ef05293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

case-2

crash info

```c
0x00007f62ab4acf15 in byte_array_to_bson_string (bytes=0x7f62a000348a "", length=length@entry=18446744073709551615) at bson_util.c:121
121     bson_util.c: No such file or directory.
(gdb) bt
#0  0x00007f62ab4acf15 in byte_array_to_bson_string (bytes=0x7f62a000348a "", length=length@entry=18446744073709551615) at bson_util.c:121
#1  0x00007f62ab4ab564 in bson_object_from_bytes_len (output=0x7f62a9097ba0, data=0x7f62a0003480 "", dataSize=16) at bson_object.c:260
#2  0x000055fb2f2e807b in protocol_handler::get_hash_id (packet=...)
    at /home/alex/Downloads/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc:1603
#3  0x000055fb2f2e85ba in protocol_handler::ProtocolHandlerImpl::HandleControlMessageEndSession (this=0x55fb303969d0, packet=...)
    at /home/alex/Downloads/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc:1628
#4  0x000055fb2f2e6ca0 in protocol_handler::ProtocolHandlerImpl::HandleControlMessage (this=0x55fb303969d0, 
    packet=std::shared_ptr<class protocol_handler::ProtocolPacket> (use count 4, weak count 0) = {...})
    at /home/alex/Downloads/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc:1562
#5  0x000055fb2f2e50ce in protocol_handler::ProtocolHandlerImpl::HandleMessage (this=0x55fb303969d0, 
    packet=std::shared_ptr<class protocol_handler::ProtocolPacket> (use count 4, weak count 0) = {...})
    at /home/alex/Downloads/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc:1471
#6  0x000055fb2f2f00d8 in protocol_handler::ProtocolHandlerImpl::Handle (this=0x55fb303969d0, message=...)
    at /home/alex/Downloads/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc:2269
#7  0x000055fb2f3132b2 in threads::MessageLoopThread<utils::PrioritizedQueue<protocol_handler::impl::RawFordMessageFromMobile> >::LoopThreadDelegate::DrainQue (this=0x55fb303851e0) at /home/alex/Downloads/sdl_core/src/components/include/utils/threads/message_loop_thread.h:201
#8  0x000055fb2f312dd1 in threads::MessageLoopThread<utils::PrioritizedQueue<protocol_handler::impl::RawFordMessageFromMobile> >::LoopThreadDelegate::threadMain (this=0x55fb303851e0) at /home/alex/Downloads/sdl_core/src/components/include/utils/threads/message_loop_thread.h:184
#9  0x000055fb2f675d1f in threads::Thread::<lambda(threads::Thread*)>::operator()(threads::Thread *) const (__closure=0x7f62a9098b61, thread=0x55fb30384250)
    at /home/alex/Downloads/sdl_core/src/components/utils/src/threads/thread_posix.cc:76
#10 0x000055fb2f676478 in threads::Thread::threadFunc (arg=0x55fb30384250)
    at /home/alex/Downloads/sdl_core/src/components/utils/src/threads/thread_posix.cc:106
#11 0x00007f62ab6f4609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#12 0x00007f62aaacf293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

```c
//bug2: length<=0
char *byte_array_to_bson_string(uint8_t *bytes, size_t length) {
  char *stringVal = malloc(sizeof(char) * (length + 1));
  
  int i = 0;
  for (i = 0; i < length; i++) {
    stringVal[i] = (char)(bytes[i] & 0xFF);
  }
  stringVal[length] = 0x00;
  return stringVal;
}
```

```c
      case TYPE_STRING:
        // Buffer length is read first
        if (remainBytes >= SIZE_INT32) {
          int32_t bufferLength = read_int32_le((uint8_t **)¤t);
          remainBytes -= SIZE_INT32;
          // Type coercion int32_5 => size_t
          if (bufferLength <= remainBytes) {
            char *stringVal = byte_array_to_bson_string((uint8_t*)current, (size_t)bufferLength - 1);
            bson_object_put_string(&obj, key, stringVal);
            free(stringVal);
            current += bufferLength;
            remainBytes -= (size_t)bufferLength;
```

case-3

if string len >512 docString overflow

```c
char *bson_object_to_string(BsonObject *obj, char *out) {
  //TODO just move the pointer rather than keep a position variable
  int position = 0;
  MapIterator iterator = emhashmap_iterator(&obj->data);
  MapEntry *current = emhashmap_iterator_next(&iterator);
  position += sprintf(out, "{ ");
  while (current != NULL) {
    BsonElement *element = (BsonElement *)current->value;
    position += sprintf(&out[position], "\"%s\":", current->key);
    switch (element->type) {
      case TYPE_DOCUMENT: {
        // docString overflow
        char docString[512];
        position += sprintf(&out[position], "%s", bson_object_to_string(bson_object_get_object(obj, current->key), docString));
        break;
      }
```
some protocol test payload:
[check_payload.zip](https://github.com/smartdevicelink/sdl_core/files/8330160/check_payload.zip)

JackLivio, 2022-03-23:
hi @alelaij What kind of app were you using to connect to sdl core?

alelaij, 2022-03-24:
I have written a python script to send payloads

jacobkeeler, 2022-03-30:
Created https://github.com/smartdevicelink/bson_c_lib/issues/58 and https://github.com/smartdevicelink/bson_c_lib/issues/59 based off of this issue. This issue will be closed when case 1 is fixed, since it is an issue within this repository

jacobkeeler, 2022-04-05:
Closed via #3896

PR #3896 (2022-04-05): Check if JSON value is a map before accessing members in SendInternalError handler

Fixes #3887 (partially, case 1)

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Send a "SendInternalError" query with an integer value in the JSON data field
```
responseHeader.setJsonData("7".getBytes(StandardCharsets.UTF_8));
```
Verify that Core does not crash.
ATF script to be added

### Summary
If a non-object value is sent in the JSON data field of a SendInternalError query, an assertion is thrown by the JSON library. This PR adds a check to avoid this asser [... cut]
closed 2022-04-05 by jacobkeeler