smartdevicelink/sdl_core #3888: HMI requests invalid VR.AddCommand during CreateInteractionChoiceSet resumption without VR part

GetmanetsIrina, 2022-03-23:
### Bug Report
HMI requests invalid `VR.AddCommand` during `CreateInteractionChoice` resumption without `VR` part

##### Precondition
1. SDL and HMI are started
2. Mobile app is registered and activated

##### Reproduction Steps
1. Mobile app requests `CreateInteractionChoiceSet` without `VR Command`
2. Perform transport reconnect  
5. Mobile app is registered with actual hash id

##### Expected Behavior
SDL does not request `VR.AddCommand` for  the `CreateInteractionChoiceSet` resumption without `VR Command`

##### Observed Behavior
SDL request invalid `VR.AddCommand` message
```
SDL -> HMI [16:36:20:329]: {"id":96,"jsonrpc":"2.0","method":"VR.AddCommand","params":{"appID":856255661,"cmdID":4,"grammarID":0,"type":"Choice","vrCommands":null}}

VRRPC.js:295 FFW.VR.AddCommandResponse
RPCClient.js:323 HMI -> SDL [16:36:20:330]: {"jsonrpc":"2.0","id":96,"error":{"code":11,"message":"Parameter 'vrCommands' does not exists!","data":{"method":"VR.AddCommand"}}}
```

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/8.1-RC1 ([25e47c7](https://github.com/smartdevicelink/sdl_core/commit/25e47c7f34f90f5393958f998c1861efa966d702))
* Testing Against: HMI: https://github.com/smartdevicelink/sdl_hmi/tree/release/5.7.0 ([cd2d250](https://github.com/smartdevicelink/sdl_hmi/commit/cd2d250cba100d1e0dd354172d2e24c0bd3449fc)), SyncProxyTester: 20210729-Android

##### Logs
[SmartDeviceLinkCore.zip](https://github.com/smartdevicelink/sdl_core/files/8333946/SmartDeviceLinkCore.zip)

LitvinenkoIra, 2022-06-14:
Luxoft team is going to fix this issue

ShobhitAd, 2022-08-04:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3938

PR #3938 (2022-08-04): Fix/Invalid VR.AddCommand sent during CreateInteractionChoiceSet resumption
Fixes #3888 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
ATF tests PR submitted for issue

### Summary
Modified creation messages for AddCommand DeleteCommand requests
Added check of vr part in VRAddCommandRequest
Added check of grammar_id in VRDeleteCommandRequest
Added unit tests

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2022-08-04 by ShobhitAd