smartdevicelink/sdl_core #3895: SDL sends BasicCommunication.PolicyUpdate(sdl_snapshot.json) twice

VjKlepikov, 2022-03-30:
### Bug Report
SDL sends BasicCommunication.PolicyUpdate(sdl_snapshot.json) twice

##### Occurrences: ~2%

##### Reproduction Steps
1. SDL is built with EXTERNAL_PROPRIETARY flag
2. PTU is triggered HMI->SDL: HMI sends SDL.OnPolicyUpdate

##### Expected Behavior
SDL->HMI: SDL.OnStatusUpdate("UPDATE_NEEDED")
SDL->HMI: BasicCommunication.PolicyUpdate(sdl_snapshot.json)

##### Observed Behavior
SDL->HMI: SDL.OnStatusUpdate("UPDATE_NEEDED")
SDL->HMI: SDL sends BasicCommunication.PolicyUpdate(sdl_snapshot.json) request twice

##### OS & Version Information
* OS/Version: Ubuntu 18
* SDL Core Version:  https://github.com/smartdevicelink/sdl_core/tree/release/8.1.0 (b30f0125) (also reproduced in 8.0.0)
* Testing Against: [What you tested with to observe this behavior. Proxy, App, HMI etc.]

##### Test Case, Sample Code, and / or Example App
./test_scripts/Policies/App_Permissions/005_ATF_DISALLOWED_app_Id_policies_And_RequestType_Validation.lua

LitvinenkoIra, 2022-06-02:
Luxoft team is going to fix this issue

ShobhitAd, 2022-06-28:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3921

closed 2022-06-28 by ShobhitAd