smartdevicelink/sdl_core #3903: SDL doesn't resume app's HMI level FULL during resumption

VjKlepikov, 2022-04-08:
### Bug Report
SDL doesn't resume app's HMI level FULL during resumption

##### Occurrences: 0.1%

##### Reproduction Steps
1. App is registered
2.  AddCommand, AddSubMenu, CreateInteractionChoiceSet, SetGlobalProperties, SubscribeButton, SubscribeVehicleData,
SubscribeWayPoints, CreateWindow, GetInteriorVehicleData (<Rpc_n>) are sent by app
3. App re-registers with actual HashId

##### Expected Behavior
SDL resumes app's HMI level FULL during resumption

##### Observed Behavior
SDL doesn't resume app's HMI level FULL during resumption

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: release/8.1.0 (6b423eb0), release/8.0.0

##### Test Case, Sample Code, and / or Example App
./test_scripts/Resumption/Handling_errors_from_HMI/017_resume_all_data_without_response_to_one_rpc_unexpected_disconnect.lua

##### Logs:
[2022-04-08_01-47-53.023_failed.tar.gz](https://github.com/smartdevicelink/sdl_core/files/8450281/2022-04-08_01-47-53.023_failed.tar.gz)

LitvinenkoIra, 2022-04-21:
Luxoft team is going to prepare a fix

ShobhitAd, 2022-06-21:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3915

closed 2022-06-21 by ShobhitAd