smartdevicelink/sdl_core #3908: Core dump by updated odometer receiving

GetmanetsIrina, 2022-04-15:
### Bug Report
Core dump by updated odometer receiving

##### Occurrences: 100%

##### Reproduction Steps
1. SDL and HMI are started
2. Mobile app is registered and activated
3. PTU is performed with [ptu_update_rc.json](https://luxproject.luxoft.com/jira/secure/attachment/1401462/1401462_ptu_update_rc.json) (with RC permissions and specified modules)
4. Mobile app requests SetInteriorVD for some module
5. Set value `3000` for odometer in "Vehicle Info" -> "Odometr"
6. PTU s triggered
7. PTU is performed with [ptu_update_rc_all_modules.json](https://luxproject.luxoft.com/jira/secure/attachment/1401118/1401118_ptu_update_rc_all_modules.json)(with RC permissions and empty array for modules)
8. Mobile app requests SetInteriorVD for some module
9. Set value `5000` for odometer in "Vehicle Info" -> "Odometr"

##### Expected Behavior
PTU is triggered, there is no core dump

##### Observed Behavior
Core dump [sdl_log_trace_proprietary.zip](https://luxproject.luxoft.com/jira/secure/attachment/1401404/1401404_sdl_log_trace_proprietary.zip) [dump_core_odometer_http.zip](https://luxproject.luxoft.com/jira/secure/attachment/1401411/1401411_dump_core_odometer_http.zip)
`DEBUG [13 Apr 2022 11:28:57,382][139958891312896][SDLMain] /home/jenkins/workspace/Release_8.1_SDL_NoUT_H/src/appMain/life_cycle_impl.cc:241 {anonymous}::sig_handler: SIGSEGV signal has been caught`

##### OS & Version Information
* OS/Version: Ubuntu 18.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/tree/release/8.1.0 (6b423eb)
* Testing Against: HMI: https://github.com/smartdevicelink/sdl_hmi/tree/release/5.7.0 (31eed2d), Mobile app: SyncProxyTester(20210729-Android)

GetmanetsIrina, 2022-04-15:
Please notice this is a non-regression issue. It's also reproduced on release 8.0

LitvinenkoIra, 2022-06-02:
Luxoft is going to fix this issue

jacobkeeler, 2022-06-22:
Closed via #3930

closed 2022-06-22 by jacobkeeler