smartdevicelink/sdl_core #3931: Core stops generating vrHelp and helpPrompt after ResetGlobalProperties

jacobkeeler, 2022-06-20:
### Bug Report
If ResetGlobalProperties is sent for VRHELP, VRHELPTITLE, and HELPPROMPT, an auto-generated value for each of these fields is sent initially, but these auto-generated values no longer update after this. 

##### Reproduction Steps
1. Register App
2. Send AddCommand, verify UI.SetGlobalProperties + TTS.SetGlobalProperties  is sent with info from this command
3. Send SetGlobalProperties with vrHelp + vrHelpTitle + helpPrompt
4. Send ResetGlobalProperties with VR_HELP, VR_HELP_TITLE, verify UI.SetGlobalProperties + TTS.SetGlobalProperties matches step 1
5. Send another AddCommand

##### Expected Behavior
UI.SetGlobalProperties  + TTS.SetGlobalProperties is sent with data for both commands

##### Observed Behavior
UI.SetGlobalProperties  + TTS.SetGlobalProperties is not sent

##### OS & Version Information
* OS/Version: Ubuntu 20.04
* SDL Core Version: master (Release 8.1)
* Testing Against: RPC Builder JS

jacobkeeler, 2022-07-01:
Seems this is because the `sending_type_` is never updated when ResetGlobalProperties is sent, see https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/help_prompt_manager_impl.cc#L340

ShobhitAd, 2022-08-04:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3937

PR #3937 (2022-08-04): Fix/Reset sending type on ResetGlobalProperties
Fixes #3931

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Submitted ATF test PR for defect

### Summary
Implements a `OnResetGlobalPropertiesReceived` method in the help_prompt_manager. When a ResetGlobalProperties request is received from an application the sending_type_ will be reset based on the global property ids in the request

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2022-08-04 by ShobhitAd